PyFPGA/pyfpga #3: Verify windows support

rodrigomelo9, 2021-02-11:


motty-mio2, 2021-02-19:
I found a problem on windows.
So I made a pull-request https://github.com/PyFPGA/pyfpga/pull/19#issue-576159749

rodrigomelo9, 2021-02-19:
Hi @motty-mio2 again thanks (merged, whit a modification to only be applied in Tcl files). It was the only change that you needed? PyFPGA works for you in Windows with Vivado?

Regards,
Rodrigo

motty-mio2, 2021-02-20:
Thank you for this merge.
With this change, I can generate bitstream file for FPGA in Windows with Vivado. Thanks.

rodrigomelo9, 2021-02-20:
Great to know :-D

I could probably add a CI action to check the python part on windows, also the [openflow](https://github.com/PyFPGA/openflow) solution (based on FOSS and solved from a bash script), but I can't check the vendor tools there. I need to add some kind of Tcl verification (such as to check the file separators).

Again, thanks to you.

gts-bzi, 2024-08-05:
Tested PR #43 by Markus on Windows (commit 65b43b8) and it seems to work when pytest is ran in pyfpga/tests.
Running `pytest tests` in project root does not seem to work as pytest ignores the `tests` directory.
Might be helpful to fix in the readme.

rodrigomelo9, 2024-08-06:
Fixed (you can run pytest also from project root now).

rodrigomelo9, 2024-09-07:
Fixed on dev

PR #55 (2024-09-07): CI: add to verify on Windows
Closes #3
closed 2024-09-07 by rodrigomelo9