PyFPGA/pyfpga #25: Load FPGA remote

qarlosalberto, 2021-04-01:
It would be really interesting to be able to load the FPGA remotely.

rodrigomelo9, 2021-04-19:
Hi @qarlosalberto sorry for the delay (I didn't receive a notification, now well configured). Could you provide me further information, maybe an example?

qarlosalberto, 2021-04-19:
Hi :)

With Vivado (I don't know if it's the same with other) is possible to have a HW Server Manager running in a remote machine. So you can load the bitstream remotely. 

Page 25: https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug908-vivado-programming-debugging.pdf

rodrigomelo9, 2021-04-19:
Ok, I know what you are talking about, and as you, not sure about other vendors. Let me see what can I do.

rodrigomelo9, 2022-05-15:
I will rewrite/simplify PyFPGA. Part of this simplification will include supporting only common scenarios of FPGA programming, not memories, not remote, etc. The aim of PyFPGA is to simplify/abstract bitstream generation, and programming is there only to quickly check the result.

closed 2022-05-15 by rodrigomelo9