PyFPGA/pyfpga #41: File paths in TCL on Windows

gts-bzi, 2024-07-05:
TCL requires paths (even on windows) to be in Posix format (forward slash over backslash).
This can easily be resolved by forcing files to be stored in posix format:

            path = Path(file).resolve().as_posix()

https://github.com/PyFPGA/pyfpga/blob/c64e9f648295ebc79386853c3d0a633e324a00ac/pyfpga/project.py#L76

Verified to be working on Windows 10 with the current version of PyFPGA.

(Am unable to create pull requests at the moment so requesting through this issue :)

rodrigomelo9, 2024-07-08:
Hi @gts-bzi, thanks! It should be fixed now, could you try it?
PS: it was a little more complex than the proposal (three methods modified; as_posix convert from Path to string)

gts-bzi, 2024-07-08:
The issue itself seems fixed, however, either this other changes since the bug report have caused the TCL file to be weirdly formatted. It works, it just has a number of extraneous newlines. Will have to check later. Generated TCL attached.
Tested with: fa84608232053ef59e84b6862954c45acb72f194
[vivado.tcl.txt](https://github.com/user-attachments/files/16122846/vivado.tcl.txt)

rodrigomelo9, 2024-07-08:
The extra newlines are due to another thing that I'm working on.
Thanks

closed 2024-07-08 by rodrigomelo9