PyFPGA/pyfpga #42: Feature request: Option to set VHDL version

gts-bzi, 2024-07-06:
Vivado requires to explicitly set the VHDL version to 2008+ if VHDL2008 features are being used in design files.
This might be solved in multiple ways: 

 - Worst but fastest: Set all files to VHDL2008 by adding following line to vivado.jinja. This will likely not break anything and support VHDL2008 in addition, but is highly bodgy. `set_property file_type {VHDL 2008} [get_files -filter {FILE_TYPE == VHDL}]`
 - Reasonably easy: Add a new function to set VHDL version explicitly for all files.
 - Most effort but probably best: Implement both setting the version for all files + per each file add command.

I will probably implement this by myself when I find the time, so let me know before you start so we can avoid duplicate work :)

rodrigomelo9, 2024-07-08:
Hi @gts-bzi 
I'm considering two things:
* Add a general-purpose `set_option` method.
* Add an `options` parameter in the `add_<HDL>` methods.
In the meantime, you could use a hook:
`prj.add_hook('precfg', 'set_property file_type {VHDL 2008} [get_files -filter {FILE_TYPE == VHDL}]')`

rodrigomelo9, 2025-03-03:
Closing, because we agreed (online meeting) it can be implemented using hooks.