PyFPGA/pyfpga #44: Feature request: Make vendor tool path configurable

gts-bzi, 2024-08-07:
The <tool>.py currently hard-codes the tool command which is generally OK but limits the use of different tool versions for certain applications. (See https://github.com/PyFPGA/pyfpga/blob/dev/pyfpga/vivado.py#L19)
Especially if a project requires a specific version of the tool, it is not currently configurable. 

Feature request: Implement function to explicitly set tool path / command through the project configuration file.
Something like `prj.set_command('<path to tool> vivado -mode batch -notrace -quiet -source')` to overwrite the standard configuration. This function can then be used to explicit configure a certain version.

rodrigomelo9, 2024-08-08:
`Especially if a project requires a specific version of the tool, it is not currently configurable.`
To specify a tool version is out of the scope. The needed tool and its particular version shall be ready to run. I tried several alternatives to deal with that in the past, and it was always very complex to use.

I don't know Diamond, but I have supported Ise, Quartus and Vivado since 2016, and Libero since 2019, and the command name never changed. If a tool in particular has this problem, I prefer to implement a "tool name finder" in the Class that supports the tool, probably based on shutil.which.

Finally, if it is needed in a very particular situation, it is always possible to do prj.conf['make_cmd']. Instead of several setters, I could document that somewhere (a "hacking" section, or similar).

gts-bzi, 2024-08-08:
I think this "hack" will do for our use case. Thanks. 
Agree that I do not see the need to implement this in upstream PyFPGA.

closed 2024-08-08 by gts-bzi