PyFPGA/pyfpga #45: Feature request: Set TCL filename to project name

gts-bzi, 2024-08-07:
TCL files for projects are currently hard coded to the name of the tool ([Example](https://github.com/PyFPGA/pyfpga/blob/cc89caf82268d7769e6c3a3717cef7f899ded4f3/pyfpga/vivado.py#L21)).

As some tools (Lattice Diamond) use the name of the TCL file to set the project name, it would be preferable to use the project name here instead of the tool name. This would make identification of project files in the tools UI easier.

Are there any reasons against this?

rodrigomelo9, 2024-08-08:
When you create a new project, you need to specify a project name:
`prj_project new -name {{ project }} -impl $impl_name -dev {{ part }}`
Is this project name ignored?
Maybe, the problem is that all the generated files use the same basename as the script? (diamond_something in this case?). Sorry, but each tool has its particularities. For example, in Ise the basename is the top-level name... I prefer to maintain the tool name as the script name for easier identification (it is needed for debugging for example).
If you have different projects, that you need to identify clearly, you should use different output directories instead of having them all together in the same directory. The tool name was selected thinking on running the same project with different tools (the original idea of the project in 2016), and again, for debugging (if you use the project name, it is overwritten).

gts-bzi, 2024-08-22:
My bad. This was actually a configuration error on my side.

closed 2024-08-22 by gts-bzi