PyFPGA/pyfpga #46: Feature request: Make loglevel configurable

gts-bzi, 2024-08-07:
Log level seems to be hardcoded to INFO at the moment. 
https://github.com/PyFPGA/pyfpga/blob/cc89caf82268d7769e6c3a3717cef7f899ded4f3/pyfpga/project.py#L48
However, it currently does not seem possible to change the log level when running pyfpga.
It would be great if one could use something like 
`python <tool>.py --loglevel=DEBUG` 
or even just have an option in the <tool>.py to configure the log level for debugging.

Any opinions before I implement something like this?

rodrigomelo9, 2024-08-08:
It is currently configurable. You can try:

```
import logging

from pyfpga.vivado import Vivado

prj = Vivado()
prj.logger.setLevel(logging.DEBUG)
prj.set_top('TOPNAME')
```
It will print:
`2024-08-07 21:07:58 - DEBUG - Executing set_top
`

I will add it in the docs.

rodrigomelo9, 2024-08-08:
Ok, I was thinking and yes, we could add set_loglevel

gts-bzi, 2024-08-08:
Provided workaround is fine by me. Thanks!

closed 2024-08-08 by gts-bzi