PyFPGA/pyfpga #47: Add support for OPTIONS

rodrigomelo9, 2024-08-11:
- [ ] add_vlog: opt parameter
- [ ] add_vhdl: opt parameter
- [ ] add_slog: opt parameter
- [ ] add_cons: opt parameter
- [ ] set_prjopts
- [ ] set_synopts
- [ ] set_paropts
- [ ] set_bitopts

svenka3, 2025-12-02:
So if I want to add the following TCL to a quartus run, you propose we do via add_slog opt parameter above? 
```tcl
set_global_assignment -name VERILOG_INPUT_VERSION SYSTEMVERILOG_2005
```

Thanks

rodrigomelo9, 2025-12-04:
Hi @svenka3 you can use the following for your use case:

`prj.add_hook('precfg', 'set_global_assignment -name VERILOG_INPUT_VERSION SYSTEMVERILOG_2005')`

More about hooks and options here: https://pyfpga.github.io/pyfpga/advanced.html